purduesigbots/pros #215: Can't compile fresh project with manjaro linux and GCC V9.2.1

lerker100, 2020-03-05:
#### Expected Behavior:
Successfully compile a new project.

#### Actual Behavior:
`prosv5 conductor new-project TestProject`
```
Name      Version    Origin
--------  ---------  -------------
kernel    3.2.1      pros-mainline
okapilib  4.0.3      pros-mainline
Building project...
make: *** [common.mk:230: bin/cold.package.elf] Error 1
Creating cold package with libc libm libpros okapilib, [ERRORS]
arm-none-eabi-g++: error: unrecognized command line option '--whole-archive'; did you mean '--whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--no-whole-archive,'; did you mean '--no-whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'; did you mean '--data-sections'?
arm-none-eabi-g++: error: unrecognized command line option '--start-group'
arm-none-eabi-g++: error: unrecognized command line option '--end-group'
```

#### Steps to reproduce:
OS: Manjaro
[Arm-none-eabi-gcc Version](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)

Install Manjaro, arm-none-eabi-gcc from the link, and pros 3.1.4
create a new project with `prosv5 conductor new-project TestProject`

#### System information:
Platform: V5
PROS Kernel Version: 
```
Name      Version    Origin
--------  ---------  -------------
kernel    3.2.1      pros-mainline
okapilib  4.0.3      pros-mainline
```

HotelCalifornia, 2020-03-05:
duplicate of #206 which is fixed in #207

closed 2020-03-05 by HotelCalifornia

Geometrically, 2020-08-16:
Hey, I am experiencing this issue on arch with pros CLI version 3.1.4. How can I make it work?