purduesigbots/pros #217: Controller screen character limit is lower than necessary

djava, 2020-04-09:
#### Expected Behavior:
You should be able to print enough to fill the screen, or whatever the VexOS limit is, if there is one

#### Actual Behavior:
There is a fairly arbitrary 15 character limit, which basically mean that unless you're printing WWWWWWWWWWWWWWW, there is more space left over in the screen.

#### System information:
Platform: V5
PROS Kernel Version: All

#### Additional Information
I don't know if there is a limit, or what the maximum number of characters you can print on the screen is, but whatever that number is is probably what PROS should let you print.  I know that VEXCode allows up to 19 characters, I don't know if that's the hard limit.
Andrew from QUEEN said that he was going to test what the limit is, you should ask him if I haven't already edited this issue to include it.

The actual change should be as simple as changing the `CONTROLLER_MAX_COLS` value in `src/devices/controller.c`

Really the only possible side effect I can think of is that it could maybe cause a bandwidth issue in the communication between the controller and the brain but I really doubt that it's significant enough to actually cause one.

nickmertin, 2020-04-09:
Tested with RMS python on latest firmware, it appears to support 19 characters, suggesting that this is the underlying VEXos limit.

djava, 2020-04-09:
It's probably worth testing it with the VexOS system calls to test that it's actually a system limitation and not just the limit that RMS and Vexcode both use

WillXuCodes, 2020-07-14:
So apparently, the line clearing function we have for the controller text never cleared consistently, as it would only clear consistently if the old text was less pixels wide than the 19 or 15 spaces used to clear the line.

This is what I mean by this, where the word being cleared here is wider than the spaces.
[image]

What needs to be up for discussion then, is that:
Would we rather have allow users to use all 19 chars? Or attempt have more consistent clearing but with 15 (or less) chars?

djava, 2020-07-14:
I'd rather just open it up, possibly with a warning in the docs

nathan-moore, 2020-07-15:
I don't know any details here Will, but I'm guessing both robot mesh and vexcode both clear the line when they do what you're doing. So either the pr is doing something wrong with regards to that, or there's a better way of clearing a line that we're missing.

WillXuCodes, 2020-07-15:
Yeah... I've heard that this also happened when we were down to 15 lines (on the current build of pros) too. I'll try to find a different method.

djava, 2020-07-15:
Anyone wanna test if vexcode controller printing does actually clear correctly in the same scenario?

@kunwarsahni01 ?

WillXuCodes, 2020-07-15:
Seems like the same thing is happening with RMS as well with their clearline function. @djava 

(From Jah | 169Z on discord)
[image]
Code used for testing:
[image]

closed 2020-07-16 by WillXuCodes