purduesigbots/pros #219: Hot and Cold

NickBlac, 2020-06-16:
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [ERRORS]
bin/auton_functions.cpp.o: In function `std::_MakeUniq<okapi::AsyncPosIntegratedController>::__single_object std::make_unique<okapi::AsyncPosIntegratedController, std::shared_ptr<okapi::AbstractMotor> const&, okapi::TimeUtil>(std::shared_ptr<okapi::AbstractMotor> const&, okapi::TimeUtil&&)':
c:\program files\pros\toolchain\usr\arm-none-eabi\include\c++\7.2.1\bits/unique_ptr.h:825: undefined reference to `okapi::AsyncPosIntegratedController::AsyncPosIntegratedController(std::shared_ptr<okapi::AbstractMotor> const&, okapi::TimeUtil const&)'
bin/auton_functions.cpp.o: In function `okapi::ChassisControllerFactory::create(std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, okapi::AbstractMotor::GearsetRatioPair, okapi::ChassisScales const&, okapi::TimeUtil const&)':
c:\program files\pros\toolchain\usr\arm-none-eabi\include\c++\7.2.1\ext/new_allocator.h:136: undefined reference to `okapi::SkidSteerModel::SkidSteerModel(std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, double, double)'
Error: Failed to build
bin/auton_functions.cpp.o: In function `okapi::ChassisControllerFactory::create(std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::AbstractMotor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, std::shared_ptr<okapi::ContinuousRotarySensor> const&, okapi::AbstractMotor::GearsetRatioPair, okapi::ChassisScales const&, okapi::TimeUtil const&)':
C:\Users\alvin\Documents\GitHub\1958-Vex-Team/./include/okapi/impl/chassis/controller/chassisControllerFactory.hpp:224: undefined reference to `okapi::ChassisControllerIntegrated::ChassisControllerIntegrated(okapi::TimeUtil const&, std::shared_ptr<okapi::ChassisModel> const&, std::unique_ptr<okapi::AsyncPosIntegratedController, std::default_delete<okapi::AsyncPosIntegratedController> >, std::unique_ptr<okapi::AsyncPosIntegratedController, std::default_delete<okapi::AsyncPosIntegratedController> >, okapi::AbstractMotor::GearsetRatioPair, okapi::ChassisScales const&)'
bin/auton_functions.cpp.o: In function `__static_initialization_and_destruction_0(int, int) [clone .constprop.92]':
C:\Users\alvin\Documents\GitHub\1958-Vex-Team/src/auton_functions.cpp:15: undefined reference to `okapi::MotorGroup::MotorGroup(std::initializer_list<okapi::Motor> const&)'
C:\Users\alvin\Documents\GitHub\1958-Vex-Team/src/auton_functions.cpp:15: undefined reference to `okapi::MotorGroup::MotorGroup(std::initializer_list<okapi::Motor> const&)'
collect2.exe: error: ld returned 1 exit status
make: *** [common.mk:240: bin/hot.package.elf] Error 1

#### Expected Behavior:

#### Actual Behavior:

#### Steps to reproduce:
IDK

#### System information:
Platform: V5
PROS Kernel Version: 3.2.1

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2020-06-16:
this looks like a problem with your code, and not a PROS-specific issue. maybe @Octogonapus wants to weigh in because it looks to be okapilib-related.

closed 2020-06-16 by HotelCalifornia

nathan-moore, 2020-06-16:
doing a clean build may help (prosv5 make clean), fyi

lerker100, 2020-06-30:
It looks like he didn't reference his motor correctly. @NickBlac if you haven't fixed it yet I'd be happy to help you on vexforum.com if you make a post with your code and the error. Tag me by the same name and I'll see it.