purduesigbots/pros #223: Better code organization

HotelCalifornia, 2020-07-10:
#### what do you mean?

I was looking over the API today and I realized that we have a lot of classes sitting in the top level `pros` namespace that can probably be grouped for clarity of purpose. The most egregious example is all the warty `ADI*` classes.

#### what do you propose?

```cpp
namespace adi {
class AnalogIn;
class AnalogOut;
class DigitalIn;
class DigitalOut;
class Encoder;
class Gyro;
class Motor;
class Port;
class Ultrasonic;
}

/** these ones are less of a concern, but could still be done */

namespace device { // also I'm not set on this name
class Controller;
class Imu;
class Motor;
class Vision;
// plus the other new devices
}

namespace rtos {
class Mutex;
class Task;
}

// pros::Serial can either stay part of the top-level namespace or get moved to a new one like pros::misc or something
```

#### what about backwards compatibility?

```cpp
#define LEGACY_TYPEDEF(old_name, new_name) \
    using old_name [[deprecated("use " #new_name " instead")]] = new_name

LEGACY_TYPEDEF(ADIAnalogIn, pros::adi::AnalogIn);

// etc
```

#### ...but why?

really I'm just not sure why we ended up going with warty names for the ADI classes only, and I think it would just be a better move to group them using the c++ method for logical groupings 😄 

#### RFC

this is a request for comments, and not a concrete plan of action by any means. I'd love for people to let me know what they think 👍

WillXuCodes, 2020-07-10:
For device we could use "hardware" to be more clear about what's in that inner layer, I feel like "devices" would refer more to the brain or controller rather than pieces of hardware.

djava, 2020-07-11:
I think that v5 might be a better name for devices, since it's all the hardware specific to the v5 system.

nickmertin, 2020-07-11:
Another option for backwards-compatibility would be to make the nested namespaces `inline`. That would everything to be accessible through the `pros` namespace, not just types.

HotelCalifornia, 2020-07-11:
> Another option for backwards-compatibility would be to make the nested namespaces `inline`.

If I understand how inline namespaces work correctly, this on its own will not be sufficient in the ADI types case at least (where the actual type names would be changing)

nickmertin, 2020-07-11:
> where the actual type names would be changing

Good point, I forgot about that. Using `inline` namespaces would require keeping the unqualified names the same.

WillXuCodes, 2020-07-11:
> I think that v5 might be a better name for devices, since it's all the hardware specific to the v5 system.
> 
> This would also help (albeit in a rare use case) differentiate from a v5 motor and a 393 motor plugged into an ADI port.

Agreed, looking back this makes more sense for smart port devices.

theol0403, 2020-07-11:
I agree with moving the ADI classes to their own namespace, but I think the core-functionality classes should stay the same. Classes like `Motor`, `Controller`, and `Task` should be treated as first-class citizens, as they are the core of PROS functionality and the main (if not only) classes most users use. Moving them behind a namespace does not really improve organization imo, they belong within the `pros` namespace as much as they do behind a `devices` or `v5` namespace, as the whole point of PROS is to interface with devices on the v5.

HotelCalifornia, 2020-07-11:
> I agree with moving the ADI classes to their own namespace, but I think the core-functionality classes should stay the same. Classes like `Motor`, `Controller`, and `Task` should be treated as [...]

I hear what you're saying. like I said in the OP, the most pressing issue in my mind is the ADI classes. as far as organizing the other classes (from a documentation standpoint), one solution could be to look at Doxygen's `\group` directive

nickmertin, 2020-07-12:
> Moving them behind a namespace does not really improve organization imo, they belong within the `pros` namespace as much as they do behind a `devices` or `v5` namespace, as the whole point of PROS is to interface with devices on the v5.

This sounds like a more appropriate scenario for `inline` namespaces.

WillXuCodes, 2020-07-14:
> This sounds like a more appropriate scenario for `inline` namespaces.

Correct me if I'm wrong, but would this also handle backwards compatibility? (Users could still type pros::class_method rather than pros::device::class_method)

WillXuCodes, 2020-07-14:
(Unless that's purely there just to print a deprecated warning)

nickmertin, 2020-07-14:
> Correct me if I'm wrong, but would this also handle backwards compatibility? (Users could still type pros::class_method rather than pros::device::class_method)

Yes it would.

> (Unless that's purely there just to print a deprecated warning)

I don't believe it would print a warning, that's not deprecated in any way. Either way, I think this is being rolled up into PROS 4 (#224), in which case I don't think there is a problem with creating breaking changes.

WillXuCodes, 2020-07-14:
> I don't believe it would print a warning, that's not deprecated in any way. Either way, I think this is being rolled up into PROS 4 (#224), in which case I don't think there is a problem with creating breaking changes.

What I meant by this was the macro function that hotel had here:
#define LEGACY_TYPEDEF(old_name, new_name) \
    using old_name [[deprecated("use " #new_name " instead")]] = new_name

LEGACY_TYPEDEF(ADIAnalogIn, pros::adi::AnalogIn);

HotelCalifornia, 2020-07-14:
the deprecated attribute will make the compiler emit a warning yes

WillXuCodes, 2020-07-14:
Right, but my question is that is this warning necessary? I feel like it'd probably just be ignored by users 90% of the time because we've made all the reorganization stuff inline.

djava, 2020-07-14:
The way I understood the original conversation was that `devices` was gonna be `inline` and `adi` wasn't. Feel free to correct me @ hotel or @ nick or someone else

WillXuCodes, 2020-07-14:
Ah right I see it, I forgot that we're removing ADI from the beginning of each adi class and need backwards compatibility with that.

WillXuCodes, 2022-06-09:
Merged in PROS 4 branch update, was not a linked issue

closed 2022-06-09 by WillXuCodes