purduesigbots/pros #227: RFC: Decouple the kernel from the SDK

TixoRebel, 2020-07-21:
#### What on earth? 
Of course the VEXOS SDK is required at some point to communicate with the V5 hardware so it's still needed somewhere, however, I'm proposing removing all direct references to it from the kernel. Instead, the kernel would link against a hardware abstraction layer (HAL) that would in turn be compiled against the VEXOS SDK.

#### Why would we want this?
Decoupling the kernel from the SDK would allow for the quicker addition of new features when the members of the PROS team is busy with other things. They would only need to update the HAL to support any new SDK features and then others could work to add and test those features in the kernel.
This would also allow for implementations of the HAL on other platforms such as a virtualized environment instead of needing to reimplement the largely undocumented SDK functions.

WillXuCodes, 2020-08-12:
Wait, doesn't this defeat the purpose of having the SDK behind a NDA? I thought VEX had the SDK behind the NDA so others wouldn't be able to create and distribute their own kernels.

What I'm trying to get at is that somebody might use the HAL to create something else non-pros related, since the implication of this is that we're allowing anybody to compile the pros kernel with the HAL.

TixoRebel, 2020-08-12:
> Wait, doesn't this defeat the purpose of having the SDK behind a NDA? I thought VEX had the SDK behind the NDA so others wouldn't be able to create and distribute their own kernels.
>
> What I'm trying [...]

I see where you're coming from but I don't think this is a problem. It's not like someone couldn't build their own kernel on top of the current PROS (or even RMS / VEXcode) kernel, you'd just be including a bunch of extra stuff that you probably don't want / need.

If someone wants to build their own kernel on top of the HAL, I don't see why we should try to stop them. Just because VEX doesn't want to publish a low level API that people can use to interface with the V5 (probably because they don't want the support calls) doesn't mean that we can't.

WillXuCodes, 2021-06-18:
With the new changes to our CI allowing for us to run it on forks, as well as our recent talks with VEX, we will not be creating a HAL.

closed 2021-06-18 by WillXuCodes