purduesigbots/pros #232: Error in building when project project directory has a '(' or ')'

grantwstaten, 2020-08-07:
#### Expected Behavior:

I expected it to build correctly as it did before I moved the project into a different folder.

#### Actual Behavior:

It gave me an error instead

#### Steps to reproduce:

move a project into a folder that has one of those characters and try to build it.

lerker100, 2020-08-08:
More information and examples would be very helpful for debugging and testing. To clarify that the problem is caused by the project folder containing the charicters '('or')'

grantwstaten, 2020-08-08:
It was the folder that the project folder was in that had a '(' and a ')', when I removed those characters from that folder's name, it then built successfully

lerker100, 2020-08-08:
A copy of the error message, what OS your running, and the folder name would be helpful.

grantwstaten, 2020-08-08:
OS was windows 10, I can't get the error message or folder name right now (I'm on my phone) , but I will comment it on here once I can

grantwstaten, 2020-08-08:
Here is the folder I did it and the error message:
.....\vex(20-21)\okapi_testing>prosv5 build
/bin/sh: -c: line 0: syntax error near unexpected token `('
Error: Failed to build
/bin/sh: -c: line 0: `if test 0 -eq 1; then printf "%s\n" "echo 'char const * const _PROS_COMPILE_TIMESTAMP = __DATE__ " " __TIME__; char const * const _PROS_COMPILE_DIRECTORY = "x(20-21)/okapi_testing";' | intercept-cc -c -x c -mcpu=cortex-a9 -mfpu=neon-fp16 -mfloat-abi=softfp -Os -g -D_POSIX_THREADS -D_UNIX98_THREAD_MUTEX_ATTRIBUTES  -Wno-psabi -ffunction-sections -fdata-sections -fdiagnostics-color -funwind-tables --std=gnu11  -o ./bin/_pros_ld_timestamp.o -"; fi;'
make: *** [common.mk:239: bin/hot.package.elf] Error 1

After changing nothing except for the folder name, if does this instead:
.....\vex20-21\okapi_testing>prosv5 build
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [OK]
Section sizes:
   text    data     bss   total     hex filename
 3.88KB   8.00B  20.00B  3.91KB     f9c bin/hot.package.elf
Creating bin/hot.package.bin for VEX EDR V5 [DONE]

WillXuCodes, 2020-08-09:
Correct me if I'm wrong, but isn't this more of a CLI issue?

Buzzec, 2020-08-10:
I think this may be able to be dealt with better over at [pros-cli](https://github.com/purduesigbots/pros-cli).

HotelCalifornia, 2021-01-23:
> I think this may be able to be dealt with better over at [pros-cli](https://github.com/purduesigbots/pros-cli).

it's actually a makefile issue. that said, our long-held position is that we're not going to bend over backwards to support strange file paths.

closed 2021-01-23 by HotelCalifornia