purduesigbots/pros #240: Error while creating/making new project.

Geometrically, 2020-08-16:
#### Expected Behavior:
The project is made and I can upload it to my robot.

#### Actual Behavior:
The command spits out this error:
` prosv5 make
Creating cold package with libc libm libpros okapilib, [ERRORS]
arm-none-eabi-g++: error: unrecognized command-line option '--whole-archive'; did you mean '--whole-file'?
make: *** [common.mk:230: bin/cold.package.elf] Error 1
arm-none-eabi-g++: error: unrecognized command-line option '--no-whole-archive,'; did you mean '--no-whole-file'?
arm-none-eabi-g++: error: unrecognized command-line option '--gc-sections'; did you mean '--data-sections'?
Error: Failed to build
arm-none-eabi-g++: error: unrecognized command-line option '--start-group'
arm-none-eabi-g++: error: unrecognized command-line option '--end-group'`

#### Steps to reproduce:
Download PROS CLI and `arm-none-eabi-g++` on Archlinux. 
Create a new project/make an existing one

#### System information:
Platform: V5
PROS Kernel Version: 3.2.1

#### Additional Information
I am on arch linux KDE

#### Screenshots/Output Dumps/Stack Traces
[image]

lerker100, 2020-08-16:
duplicate of #206 which is fixed in #207. Just change the files that #207 changed and compile.

Geometrically, 2020-08-16:
Where are those files located?

Geometrically, 2020-08-16:
Nevermind, found it, sorry. When I recompile, this happens:
[image]

nathan-moore, 2020-08-16:
Try it with these changes: https://github.com/purduesigbots/pros/pull/233/files

Geometrically, 2020-08-16:
I got another error. How can I change it so it uses the updated file when I create a new project?

nathan-moore, 2020-08-16:
What error @Geometrically?

lerker100, 2020-08-17:
Wait until pros releases the next version with the patch. Or just fix every project you make

nathan-moore, 2020-08-17:
I can give you a template, or we can push an update. You mentioned another error @Geometrically. Was it the first one you mentioned then?

Geometrically, 2020-08-20:
Here is the new error:
```
make: *** [common.mk:229: bin/cold.package.elf] Error 1
Creating cold package with libc,libm,libpros,okapilib [ERRORS]
Error: Failed to build
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:./firmware/v5-common.ld:14: warning: redeclaration of memory region `COLD_MEMORY'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:./firmware/v5-common.ld:15: warning: redeclaration of memory region `HEAP'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld:./firmware/v5-common.ld:16: warning: redeclaration of memory region `HOT_MEMORY'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lc
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status```

Could you push an update?

nathan-moore, 2020-08-20:
> Here is the new error:
> make: *** [common.mk:229: bin/cold.package.elf] Error 1
> Creating cold package with libc,libm,libpros,okapilib [ERRORS]
> Error: Failed to build [...]

And that's with the changes from https://github.com/purduesigbots/pros/pull/233/files? Can you double check?

> Could you push an update?

I mean, first we'd want to make sure this actually fixes the issues you're having.

nathan-moore, 2020-08-21:
A thought: the root cause here is because you don't have newlib. You don't need to link against it, but you will need the headers.

HotelCalifornia, 2020-08-21:
if you're on arch linux you should be able to install arm-none-eabi-newlib