purduesigbots/pros #252: pros::IMU : How to reset rotation angle, yaw and heading without calibrating the IMU. There are no methods reset_rotation(), reset_heading() etc.

AVMS1, 2020-09-18:
#### Expected Behavior:
set IMU sensor's rotation, yaw and heading to zero value; currently IMU retains the value from the prev reading. 

#### Actual Behavior:
I would like to reset IMU sensor's rotation to zero prior to every turn and don't see method to reset_rotation() like robotmesh api. Are there any methods available to reset heading, rotation etc. in the IMU without actually calibrating the IMU prior to each turn or movement...

#### Steps to reproduce:
none

#### System information:
Platform: V5
PROS Kernel Version:  3.2.1

#### Additional Information
none

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2020-09-18:
There is not currently a way to do this with the PROS API, but we should add it (cc @kunwarsahni01 / @WillXuCodes). in the meantime, you can make a class that wraps or inherits from the `pros::Imu` class with this functionality

AVMS1, 2020-09-18:
> There is not currently a way to do this with the PROS API, but we should add it (cc @kunwarsahni01 / @WillXuCodes). in the meantime, you can make a class that wraps or inherits from the `pros::Imu` class with this functionality

Thank you!

AVMS1, 2020-09-22:
I may be missing something here... I have implemented pid functions to turn and move the chassis using pros:Imu. If there is no function, say to reset the rotation, how can I make two turns in autonomous code say 45 degrees and another 90 degrees? Should I 'maintain' previous rotation value and manipulate it for the next?

WillXuCodes, 2020-09-22:
> I may be missing something here... I have implemented pid functions to turn and move the chassis using pros:Imu. If there is no function, say to reset the rotation, how can I make two turns in [...]

You could try saving the current angle and add it to how many x degrees you want to turn in the function where you turn another x degrees. 

Also, I recommend vexforum or the VTOW discord server for questions about coding advice. This github page is for discussion about bugs and features to be implemented. We thank you greatly for the idea of this feature :).

PR #260 (2021-01-23): ✨ Add Zeroing and Position Setting Functions for the IMU
#### Summary:
Added zeroing and current value setting for readings for the IMU.

#### Motivation:
Requested via issue, might as well add this feature as it would be beneficial for writing autons.

#### References:
closes #252 

#### Test Plan:

- [x] Test Reset Functions (C API)
- [x] Test Reset Functions (C++ API)
- [x] Test Value Set Functions (C API)
- [x] Test Value Set Functions (C ++API)
closed 2021-01-23 by HotelCalifornia