purduesigbots/pros #254: Support new color and distance sensors

dcieslak19973, 2020-09-27:
#### Expected Behavior:
Distance sensor must be able to return:
* Distance

Distance sensor should be able to return:
* Size
* Velocity

Color sensor must be able to set:
* LED Brightness

Color sensor must be able to return:
* Distance (Brain UI lists it as Near/Close/Far/None)
* Color of object

Color sensor could be able to:
* Support Gesture events (Object Up, Object Down, Object Left, and Object Right)

#### Actual Behavior:
N/A

#### Steps to reproduce:
N/A

#### System information:
Platform: V5
PROS Kernel Version: N/A

#### Additional Information
N/A

#### Screenshots/Output Dumps/Stack Traces

nathan-moore, 2020-09-27:
You'll be interested in the https://github.com/purduesigbots/pros/tree/feature/distance-sensor and https://github.com/purduesigbots/pros/tree/feature/optical-sensor branches.

dcieslak19973, 2020-09-28:
Already being worked on

nathan-moore, 2020-09-28:
Might as well keep this open until its done.

WillXuCodes, 2020-09-28:
Other than these basic changes, do you have anything else in mind that you'd potentially like as functions related to the distance and color sensors? @dcieslak19973

dcieslak19973, 2020-09-28:
Not beyond basic parity with what the Brain UI provides or Vex Code. The new headers in the feature branches look reasonable.

dcieslak19973, 2020-10-09:
@HotelCalifornia  / @WillXuCodes / @kunwarsahni01 / @mooreBrendan - are PRs

https://github.com/purduesigbots/pros/pull/258
https://github.com/purduesigbots/pros/pull/257

Able to be merged?

kunwarsahni01, 2020-10-09:
The PRs are still being reviewed and all three PRS (optical, rotation, and distance) will be merged once they are completed.

HotelCalifornia, 2020-10-28:
closed in #257 and #258

closed 2020-10-28 by HotelCalifornia