purduesigbots/pros #266: Easier github support

SizzlinSeal, 2020-11-30:
So, just to make it easier to upload the files to a github. Basically sometimes my team helps me with some of the programming and just uploading to github would be easier if there was an integrated github feature like in CLion. It would just make it faster than by copy/pasting files and stuff using github desktop.

nathan-moore, 2020-11-30:
This isn't a feature for the pros team. You can look into atom support for git, but if memory serves its pretty decent.

> github support

From what you're saying, you want git support. 

> copy/pasting files

If you're doing that, you're probably using git wrong. Look up a git tutorial or ask in the vex discord server.

closed 2020-11-30 by nathan-moore

WillXuCodes, 2020-11-30:
Hi SizzlinSeal,

For editor features, these would go under our PROS editor repository. As for your concerns with github integration with the editor, we will take it into consideration as a potential feature in the future when we transition from VSCode to Atom. 

I've also noticed that your team has been copying and pasting into github desktop, which is not a recommended way of using git. I recommend taking a look at this version control article on our wiki for a better understanding as to how to use git.
https://wiki.purduesigbots.com/software/general/version-control

Udit8348, 2020-12-02:
@SizzlinSeal Someone had a similar question a few days ago and I made a clear guide [here](https://docs.google.com/document/d/1T3M5Vsdz_Rr3cwiMe_vhnbjNMS_JDnZg2Wt2hHtqo5Q/edit#heading=h.bjs69tb92zrq). I recommend you check this out, and also follow the advice given above too. Try setting up the workflow on a blank project before your actual project. You can tag me on the Unofficial Vex discord at `Udit#4398` so we avoid making further comments on this closed issue.