purduesigbots/pros #284: Atom is outdated and causes packages installed by default with PROS to throw errors when updated

SizzlinSeal, 2021-03-14:
#### Expected Behavior:
For files in the file tree to have icons based on what type of files they are

#### Actual Behavior:
All files have the same icon

#### Steps to reproduce:
Download PROS and update file-icons package

#### System information:
Platform: V5
PROS Kernel Version: 3.4.0

#### Additional Information
updating atom may help, but you can't do that because of PROS, so it might have to be updated within the PROS source code

#### Screenshots/Output Dumps/Stack Traces
stack trace: https://pastebin.com/hrpM58hr
[image]

HotelCalifornia, 2021-03-14:
As a workaround, you can use Atom and install the pros-atom3 plugin plus its dependencies

SizzlinSeal, 2021-03-14:
ok thanks ill try that