purduesigbots/pros #288: Add Alias for pros::IMU as pros::Imu

WillXuCodes, 2021-05-12:
Issue is what it says on label

ad101-lab, 2021-05-12:
IMU is an acronym for

I - Inertial
M - Measurement
U - Unit

I don't think it would make sense to use the lowercase version, I think that this could just create more confusion.

WillXuCodes, 2021-05-12:
> IMU is an acronym for
> 
> I - Inertial
> M - Measurement
> U - Unit
> 
> I don't think it would make sense to use the lowercase version, I think that this could just create more confusion.

I understand what IMU stands for, it's just that sometimes people ask why their "pros::Imu" is not working and it's a pretty simple fix. However, I'm not sure what you mean by confusion as this isn't intended as a replacement for the fully uppercase type but rather just another way to do the same thing.

HotelCalifornia, 2021-05-12:
> IMU is an acronym for
> 
> I - Inertial
> M - Measurement
> U - Unit
> 
> I don't think it would make sense to use the lowercase version, I think that this could just create more confusion.

We're not going to change the current one entirely because that would be a breaking API change.

baylessj, 2021-05-14:
This reminds me of a section in the Effective Dart styleguide: https://dart.dev/guides/language/effective-dart/style#do-capitalize-acronyms-and-abbreviations-longer-than-two-letters-like-words

The addition of this alias is good by me.

commit cc0bc122 (2021-05-17): Move alias statement, remove c implementation

Move the alias statement in imu.hpp outside of IMU class. Remove c
implementation of alias.

Resolves #288
PR #293 (2021-05-17): 🚸 add pros::IMU alias
#### Summary:
Create the pros::Imu alias for pros::IMU in imu.h and imu.hpp

#### Motivation:
Reduce confusion with newer users attempting to use pros::Imu instead of pros;:IMU

##### References (optional):
closes #288
commit 07b0121b (2021-05-17): 🚸 add pros::IMU alias (#293)

* Create alias for Imu in IMU.h and IMU.hpp

* Revert version changes

* Fix version number and add typdef to pros namespace

* Revert typdef location to inside class

* Move alias statement, remove c implementation

Move the alias statement in imu.hpp outside of IMU class. Remove c
implementation of alias.

Resolves #288

* Add newline in version

* Fix indent of alias
closed 2021-05-17 by HotelCalifornia