purduesigbots/pros #298: Add GPS/VEXLink support

WillXuCodes, 2021-06-17:
Our plans are to add GPS and VEXLink support in the next update. This will happen after the SDK release (whenever that is) and _hopefully_ it will include GPS and VEXLink support. 

Other than supporting the default sensor functionality, here's what else we need to do on top of that:

### GPS TODO:
- Redefine how losing lock on the sensor is signaled to the user (vexOS calls currently _may_ not do this in a preferrable way).
- Add support for reading and feeding heading values in an atan2 output format like way. (The current system is with 0 degrees at north, going clockwise, some teams may prefer 0 degrees on the pos X axis with degrees going CCW for quadrants 1 and 2, and negative degrees for quadrants 3 and 4). 
- Feeding heading could be done with a piecewise function shown below: 
`vexStyleHeading = (heading > -180 && heading <= 90) ? (-heading + 90) : (-heading + 450);`
- Reading heading could be done by reversing the process.
(note that although mod could be used to do this, it doesn't support doubles so these calculations are used to preserve accuracy).
- We could also add other helper functions for calculating distance to point, angle to point, and other values that may be helpful for AI teams.

### VEXLink TODO:
- Given that the SDK _most likely_ will only support raw serial communication (as opposed to a built in protocol for sending messages out of the box), we will need to create our own schema and communication protocol whenever this rolls around.

nickmertin, 2021-06-17:
> (note that although mod could be used to do this, it doesn't support doubles so these calculations are used to preserve accuracy)

Is `fmod` not a feasible option?

> Given that the SDK _most likely_ will only support raw serial communication (as opposed to a built in protocol for sending messages out of the box), we will need to create our own schema and communication protocol whenever this rolls around.

True raw serial, or would it at least provide framing? (If not, we can obviously use COBS)

WillXuCodes, 2021-06-18:
> Is fmod not a feasible option?

No I just forgot it existed for a quick sec there :)

> True raw serial, or would it at least provide framing? (If not, we can obviously use COBS)

No clue, but I believe it _most likely_ will at least provide some framing in the form of functions to call to the radio.

WillXuCodes, 2021-06-22:
Due to how big the Screen API addition is, this has been moved to a separate update from 3.5.0 so we can fix bugs found in the screen api during the same phase this is in development.

bwees, 2022-01-16:
What is the timeline on Vexlink support? My team wants to switch to Pros but this is the only thing that is preventing us from switching.

WillXuCodes, 2022-01-16:
> What is the timeline on Vexlink support? My team wants to switch to Pros but this is the only thing that is preventing us from switching.

We're working on it, semester just begun but we're going at it one step at a time. Hopefully late January/early Feb we'll have a Kernel release out (version 3.6)

closed 2022-06-18 by WillXuCodes