purduesigbots/pros #312: Conductor

johnnymcmike, 2021-08-30:
#### Expected Behavior:
A new project is created as per 
#### Actual Behavior:
Command runs with this output:
```
Updating pros-mainline... Done
Applying kernel@3.5.2 [####################################] 100%
Finished applying kernel@3.5.2 to Test
Applying okapilib@4.2.0 [####################################] 100%
Finished applying okapilib@4.2.0 to Test
New PROS Project was created:
PROS Project for v5 at: /home/johnnymcmike/Documents/projects/pros/Test (Test)
Name      Version    Origin
--------  ---------  -------------
kernel    3.5.2      pros-mainline
okapilib  4.2.0      pros-mainline
Building project...
Creating cold package with libc,libm,libpros,okapilib [OK]
Stripping cold package  [DONE]
Section sizes:
   text	   data	    bss	  total	    hex	filename
1.29MB  4.35KB  47.66MB  48.95MB 30f12b2 bin/cold.package.elf
Compiled src/main.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and libc,libm,libpros,okapilib [ERRORS]
/home/johnnymcmike/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
collect2: error: ld returned 1 exit status
make: *** [common.mk:239: bin/hot.package.elf] Error 1
```
The `Test` directory is still created and is still populated. Changing the argument from `-nostartfiles` to `--nostartfiles` in the `home/johnnymcmike/.config/pros/templates/kernel@3.5.2/` causes it to not recognize `--nostartfiles` as an argument, and a similar error is produced.

#### System information:
Ubuntu 21.04, arm-none-eabi-gcc version is 10.3.1

PROS Version: 3.2.2

johnnymcmike, 2021-08-30:
That's weird. Would it be feasible to run Pros and do all of the development for it in virtual machine or WSL? I'm new to this and am setting it up for the first time.

nathan-moore, 2021-08-30:
@GLMONTER do you test with different gcc or arm-none-eabi-gcc versions? The later is what matters, and people have  temporary worked around this by rolling back an arm-none-eabi-gcc version. Did this not work for you?

johnnymcmike, 2021-08-30:
Which arm-none-eabi-gcc version should I roll back to if I want to try this workaround?

WillXuCodes, 2021-08-30:
> Which arm-none-eabi-gcc version should I roll back to if I want to try this workaround?

2020 Q2 works, it's the one running on my machine right now.

johnnymcmike, 2021-08-30:
Reverted versions and it works now. Thanks!