purduesigbots/pros #342: scrapping twittes from twitter

Tahirullah7710, 2022-01-15:
#### Expected Behavior:    this is the code behind scene....
# application variables
user = input('username: ')
my_password = getpass('Password: ')
search_term = input('search term: ')

# create instance of web driver
options = EdgeOptions()
options.use_chromium = True
driver = Edge(options=options)

# navigate to login screen
driver.get('https://www.twitter.com/login')
driver.maximize_window()
sleep(5)
username = driver.find_element_by_xpath('//input[@name="text"]')
username.send_keys(user)
username.send_keys(Keys.RETURN)
sleep(3)

password = driver.find_element_by_xpath('//input[@name="password"]')
password.send_keys(my_password)
password.send_keys(Keys.RETURN)
sleep(3)

# find search input and search for term
search_input = driver.find_element_by_xpath('//input[@aria-label="Search query"]')
search_input.send_keys(search_term)
search_input.send_keys(Keys.RETURN)
sleep(1)

# navigate to historical 'latest' tab
driver.find_element_by_link_text('Latest').click()

#### Actual Behavior: this is the ISSUE the OS occurs during run time to crapping the twittes from twitter......

OSError: [WinError 216] This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher

#### Additional Information
so please guide me how can i fix this kind of issure

HotelCalifornia, 2022-01-15:
I think you have the wrong library

closed 2022-01-15 by HotelCalifornia

Tahirullah7710, 2022-01-15:
> I think you have the wrong library
from getpass import getpass
from time import sleep
from selenium.webdriver.common.keys import Keys
from selenium.common.exceptions import NoSuchElementException
from msedge.selenium_tools import Edge, EdgeOptions
(THis is the modules and libraries )

HotelCalifornia, 2022-01-15:
PROS is a C++ library for robotics.... I'm not sure how you got here but the code you're writing has nothing to do with this repository