purduesigbots/pros #349: Reversing a single pros::Motor causes other pros::Motor objects on the same port to be reversed as well

coderbot16, 2022-03-04:
Thanks for developing this library. Overall it's been a pleasant experience to use it, but today I spent a few hours debugging some extremely confusing behavior, initially seeming like a miscompile or undefined behavior, that later turned out to be caused by some unintuitive handling of motor reversing in the C++ API.

#### Expected Behavior:

I would expect for pros::Motor to act more like okapi::Motor in this regard, where reversing a Motor object only reverses that object. I wouldn't expect from the documentation that the "reversed" property is a property of the port the motor is connected to.

#### Actual Behavior:

Reversing one pros::Motor object causes movements on other pros::Motor objects on the same port to be reversed.

#### Steps to reproduce:

Due to mixing usages of pros code (opcontrol) and okapi (auton), I ended up in a situation where I had two different objects controlling the same port. While in my case I had a pros::Motor and an okapi::Motor, this can be reproduced by using two pros::Motor objects. Just create two pros::Motor objects, reverse one of them, and note that the other one now has reversed movement.

#### System information:

Platform: V5
PROS Kernel Version: (not needed)

#### Additional Information

I found out this behavior by inspecting the current revision of the code on GitHub, which is currently https://github.com/purduesigbots/pros/commit/467264d15f437f898f6d7630aed64baf79e979b4.

Root cause - setting global state on a motor from a function that doesn't immediately suggest that it changes global state:

https://github.com/purduesigbots/pros/blob/c2dafd2d9d007f8d22e09b7de3a9d38ad88575c4/src/devices/vdml_motors.c#L327

#### Screenshots/Output Dumps/Stack Traces

Not applicable.

HotelCalifornia, 2022-03-04:
I suppose it could be clarified in the documentation that this is the behavior (as it is with similar functions acting on almost every device in the API). the current iteration of the PROS API was designed with the C API in mind first and the C++ one implemented on top of that.

If I were to go back to 2017 when we designed PROS 3 I probably would advocate even more strongly for a C++-first API than I did back then. However for now, this is what we have to deal with and changing fundamental behavior at this stage would be a major regression.

I would suggest that mixing and matching the PROS and Okapi APIs is, while technically supported, not really recommended for this reason.

WillXuCodes, 2022-06-09:
I'm going to close this issue and reference it in a new one for our plans on how we're thinking about addressing this.

closed 2022-06-09 by WillXuCodes