purduesigbots/pros #353: RFC: Change Various Structs/Enums to pros namespace, instead of being in pros::c

WillXuCodes, 2022-04-05:
Currently, we have many structs and enums (namely in imu.h and gps.h) that should probably be in the pros namespace rather than the pros::c namespace. This is to have a better distinction between what is part of the "pros::c" API vs default C++ pros API. Since the PROS API is primarily C++, we want to minimize interaction with the ::c namespace for users of the C++ only API. 

However, I'd like to get some input on this tbh. May or may not be a good change and it's definently something we want as a PROS 4 thing to prevent user confusion on PROS 3. 

Good representation of what I'm talking about
https://github.com/purduesigbots/pros-docs/pull/268

HotelCalifornia, 2022-04-05:
as long as they're defined with C linkage, POD types are fine to define in C++ and use in C. so depending on what level of support is wanted for a low level C API, we have some options.

if we want to go full on C++ it doesn't matter we can just do whatever. probably a better move would be to define two levels of structure for each logical unit, where a POD type contains the relevant data to call down to the C API and a more complex type that has all the member functions etc.

this is sort of what we do now but it would make the relationship more formal. I'm thinking something like ```cpp
namespace pros {
namespace impl {

extern "C" struct motor_data {
    std::uint8_t port;
    // whatever else idk
}

} // namespace impl

class Motor : impl::motor_data {
    // ...
}

} // namespace pros
```

WillXuCodes, 2022-06-05:
Also, for PROS 4 we should start transitioning enums into enum classes to be a more C++ focused API.

PR #390 (2022-07-25): ✨ PROS 4: Enum Classes for GPS and IMU
#### Summary:

I have moved the structs in pros::c to pros namespace in both the imu.h and gps.h. The respective modifications in the files vdml_imu.cpp and vdml_gps.cpp have also been made to appropriately match the modified namespace. I have also added an enum class and modified the method in imu.cpp to static cast the enum class.

#### Motivation:

These changes make PROS4 a more C++ focused API.

##### References (optional):

closes #353 

#### Test Plan:

Run CI tests to ensure there are no breaking changes
- [ ] test item
closed 2022-10-04 by Richard-Stump