purduesigbots/pros #362: undefined reference to getentropy

mattfbacon, 2022-05-17:
#### Expected Behavior:
Project's cold library compiles and links properly

#### Actual Behavior:
Error at link time:

```
Creating cold package with libc,libm,libpros,okapilib [ERRORS]
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/lib/thumb/v7+fp/softfp/libstdc++.a(random.o): in function `std::(anonymous namespace)::__libc_getentropy(void*)':
/build/arm-none-eabi-gcc/src/gcc-12.1.0/libstdc++-v3/src/c++11/random.cc:179: undefined reference to `getentropy'
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/lib/thumb/v7+fp/softfp/libstdc++.a(random.o): in function `std::random_device::_M_init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/build/arm-none-eabi-gcc/src/gcc-12.1.0/libstdc++-v3/src/c++11/random.cc:452: undefined reference to `getentropy'
make: *** [common.mk:229: bin/cold.package.elf] Error 1
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: ./firmware/libc.a(lib_a-arc4random.o): in function `_rs_stir':
arc4random.c:(.text._rs_stir+0x8): undefined reference to `getentropy'
collect2: error: ld returned 1 exit status
```

#### Steps to reproduce:

100% reproducible. Create a new project with `prosv5 conductor new-project` and build it with `prosv5 make`.

#### System information:
Platform: V5
PROS Kernel Version: 3.6.0

#### Additional Information

OS: Arch Linux

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2022-05-18:
yeah this happens because arch updates the toolchain faster than arm does (and also we haven't updated our copy of libc.a in like 3 years).

the workaround for you is to define

```c
int getentropy(void *buffer, size_t length) {
    return -1;
}
```

in a C file (or C++ file provided you use `extern "C"`) that is **added to the cold package** of your project. documentation for adding stuff to the cold package can be found on our site, or you can disable hot/cold linking and use a monolith build if you don't want to do that (you'll still need to provide the above definition)

@purduesigbots/pros-development can someone open an issue about updating libc.a if one hasn't been already? we'll probably need to also provide the getentropy stub even after the update because the latest official toolchain we have is 11 something and arch is shipping 12 now

mattfbacon, 2022-05-18:
OK, maybe add this as a weak symbol to libpros? Anyway thanks for the quick reply and the workaround.

HotelCalifornia, 2022-05-18:
it should be a weak symbol defined in newlib-libc but again we haven't updated ours in 3 years and the arch distribution uses a newer one than we have access to from official sources anyway, so we will just define this stub strongly in the kernel as soon as someone gets to it (this has been our strategy for solving these issues..... uhm.... since we last updated libc lol)

closed 2022-05-22 by WillXuCodes