purduesigbots/pros #364: Linker script syntax error

mattfbacon, 2022-05-26:
OS: Arch Linux
PROS version: 3.6.0

When running `prosv5 make` and/or `make`:

```
Creating cold package with libc,libm,libpros,okapilib [ERRORS]
/usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld:./firmware/v5-common.ld:30: syntax error
collect2: error: ld returned 1 exit status
make: *** [common.mk:229: bin/cold.package.elf] Error 1
```

The syntax looks fine to me. Not sure what's tripping up the linker here.

mattfbacon, 2022-05-26:
I would assume it's an issue with ld itself since the linker script was last updated two years ago.

HotelCalifornia, 2022-05-26:
my comment from the previous issue still stands: does this occur on older versions of the toolchain?

also minor nit, the linker script the error mentions was updated ~5 months ago, when you noticed and fixed a typo in one of the section names

mattfbacon, 2022-05-26:
> was updated ~5 months ago,

True, though the line it mentions was updated 2 years ago per the blame. Touche.

> does this occur on older versions of the toolchain?

I don't have any easy way to access older versions of the toolchain on Arch Linux. I would probably have to build from source. What version do you have on your machine?

HotelCalifornia, 2022-05-26:
Latest official distribution (er, last official distribution before the generic arm-gcc release) is 10.3:

https://developer.arm.com/downloads/-/gnu-rm

HotelCalifornia, 2022-06-01:
did you try using an older version of the toolchain like I suggested? There's a Linux distribution at the link I provided above

mattfbacon, 2022-06-01:
Yeah it works with the linker in that package. I also tested with the latest toolchain on <https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads>, which is version 2.37. So I suppose there is a regression in 2.38.

HotelCalifornia, 2022-06-01:
i truly have no idea what's going on here or what changed. 

> which is version 2.37. So I suppose there is a regression in 2.38.

what is?

also i guess for completeness's sake can you post the full v5-common.ld you're getting in your projects?

mattfbacon, 2022-06-01:
Okay, I'll test on different distros a bit to see if it's Arch-specific. Thanks for the guidance. It helps more than you might think to get help from someone that has a working knowledge of the library. (`common.mk` is gibberish to me :sweat_smile: )

HotelCalifornia, 2022-06-01:
ok i'm stupid as usual because that was you also never mind

mattfbacon, 2022-06-01:
Fedora has 2.37. Trying Ubuntu.

HotelCalifornia, 2022-06-01:
don't bother with Ubuntu it's waaaay behind and to my knowledge doesn't have any official distribution beyond what's on the arm site

HotelCalifornia, 2022-06-13:
Ok i managed to figure out the issue:

looks like between binutils 2.33 and 2.34 `MEMORY` was added as a specific command, which means our section named `MEMORY` is being parsed as a keyword and not as a section name. this should be trivial to fix, just need to rename that section in all the linker scripts.

HotelCalifornia, 2022-06-13:
actually i take it back. i'm not sure how this worked at all given that `MEMORY` has always been a command. in any case the `REGION_ALIAS` in v5-hot.ld and v5.ld just needs to define a region other than the reserved word MEMORY

PR #376 (2022-06-15): 🐛 fix linker script improper use of keyword as region name
Gnu ld 2.38 now uses MEMORY as a keyword, while the pros linker scripts uses it as the name of a region, causing an error during linking that prevents any project from building. This PR contains a simple fix for that by renaming all instances of MEMORY to RAM. 

closes #364
closed 2022-06-15 by WillXuCodes