purduesigbots/pros #375: Update to C++20

karmanyaahm, 2022-06-10:
Hello,
Are there any plans to update default projects to C++20 rather than 17?  I'm doing an OkapiLib feature, and it would be really nice to have everyone using c++20 for std::format

HotelCalifornia, 2022-06-10:
not currently but i've documented a way to change it here:

https://github.com/hotelcalifornia/libhotel#usage

this would also be a very easy feature to PR here, if you want to take a stab at it. I'd been meaning to add this but never got around to it

HotelCalifornia, 2022-06-10:
also looking at the readme again i think the if statements can be replaced with ```mk
C_STANDARD?=gnu11
CXX_STANDARD?=gnu++17
```

HotelCalifornia, 2022-06-10:
> Are there any plans to update default projects to C++20 rather than 17?

lastly, to answer this specifically: probably not until PROS 4

WillXuCodes, 2022-06-15:
Yeah we could totally do this for PROS 4. I'll work on it in the coming few days.

WillXuCodes, 2022-06-24:
> using c++20 for std::format

Sorry to break it to you, but I don't think GCC supports std::format in any form at the moment. Although we will be upgrading for PROS 4, I recommend you look for other options for whatever you are trying to accomplish.

karmanyaahm, 2022-06-24:
Yes, thanks for taking care of this. I have already implemented this using sprintf and a fixed buffer
 https://github.com/OkapiLib/OkapiLib/pull/487/files#diff-cd6a0e232dbe657c242a1f904fc3c7f1b9e03f1811e4d9c77c7128d774cec54bR18-R32

WillXuCodes, 2022-08-02:
This is done for PROS 4.

closed 2022-08-02 by WillXuCodes