purduesigbots/pros #398: imu_reset appears to normally time out

Octogonapus, 2022-08-01:
#### Expected Behavior:
`pros::c::imu_reset` should wait until the IMU has set its calibration flag. This calibration process takes approximately 2 seconds. The `imu_reset` function should also time out after an excessive amount of time has passed, though this should not be the normal operating mode.

#### Actual Behavior:
The `imu_reset` function appears to normally time out. It has a 1 second timeout, and the IMU calibration process is documented to take around 2 seconds, so this is logical.

It would be helpful if someone with an IMU could create a MWE that simply resets the IMU using the VDML directly and measures how long the reset takes.

#### System information:

latest V5 kernel

Octogonapus, 2022-08-01:
Once someone with an IMU can confirm the true calibration time, we can consider adjusting the `imu_reset` timeout.

WillXuCodes, 2022-08-01:
So the code in question I believe is this: 
```C++
do {
		port_mutex_give(port - 1);
		task_delay(5);
		timeoutCount += 5;
		claim_port_i(port - 1, E_DEVICE_IMU);
		if (timeoutCount >= IMU_RESET_TIMEOUT) {
			port_mutex_give(port - 1);
			errno = EAGAIN;
			return PROS_ERR;
		}
		device = device; // suppressing compiler warning
	} while(!(vexDeviceImuStatusGet(device->device_info) & E_IMU_STATUS_CALIBRATING));
```

This isn't intended to be a blocking call to wait until the IMU is done calibrating, but to wait until the hardware is finished setting the "calibrating" flag. This was to prevent an issue where the IMU wasn't doing this correctly and checking the IMU status immediately after resetting was leading to the status being reported incorrectly as "not calibrating". 

However, it's really strange that it's not setting the flag now... we should test this with hardware and figure out what's going on.

WillXuCodes, 2022-08-01:
TLDR: 1 second is not how long we wait for the IMU to fully reset, but rather how long we are waiting for the IMU/hardware to set the status flag correctly.

Octogonapus, 2022-08-01:
> So the code in question I believe is this:

Indeed, it is.

> This isn't intended to be a blocking call to wait until the IMU is done calibrating, but to wait until the hardware is finished setting the "calibrating" flag.

Oh, my understanding was that those two are the same thing. What is the proper way to check if calibration has completed?

WillXuCodes, 2022-08-01:
> > This isn't intended to be a blocking call to wait until the IMU is done calibrating, but to wait until the hardware is finished setting the "calibrating" flag.
> 
> Oh, my understanding was that those two are the same thing. What is the proper way to check if calibration has completed?

With the code above, what *should* happen is that you can just call imu.get_status() immediately afterwards to check the status and it should be correct... are you encountering something different occuring?

Octogonapus, 2022-08-01:
This issue stems from debugging someone's problem on Discord in which they had to add an additional delay after `reset` finished to get good data from `get_rotation`.

Looking at the C++ docs in the source, part of this confusion is because the documentation in the header files and the docs website is inconsistent and part of it is because the documentation on the website is unclear. The reset implementation actually looks correct to me.

I guess there are two things, then:
1. We can update the documentation phrasing for clarity
2. Once the calibration flag is set, more time is required to finish calibration. Does vexOS give us any way to determine if the IMU has finished its calibration routine?

WillXuCodes, 2022-08-01:
> This issue stems from debugging someone's problem on Discord in which they had to add an additional delay after `reset` finished to get good data from `get_rotation`.
>
> Looking at the C++ docs in the [...]

I think the best way to do this is to add clarifying documentation that the canonical way to ensure that the IMU resets properly with a delay is to do something like this:

while(imu_sensor.get_status() == E_IMU_STATUS_CALIBRATING) pros::delay(10);

We'd probably want to put it under https://pros.cs.purdue.edu/v5/api/cpp/imu.html?highlight=imu%20reset#reset. By design, we don't want this call to be blocking and we'd rather leave it in the hands of the user to implement this. 

Another option that we could also do is add an `imu_reset_blocking` call that does this delay for them. This would also make writing code in PROS slightly easier.

Octogonapus, 2022-08-01:
> I think the best way to do this is to add clarifying documentation that the canonical way to ensure that the IMU resets properly with a delay is to do something like this:
> while(imu_sensor.get_status() == E_IMU_STATUS_CALIBRATING) pros::delay(10);

Oh it sets that flag during calibration, and clears it when calibration is finished.

> By design, we don't want this call to be blocking and we'd rather leave it in the hands of the user to implement this.

Great, I agree.

> Another option that we could also do is add an `imu_reset_blocking` call that does this delay for them. This would also make writing code in PROS slightly easier.

That is perhaps worth doing. We see this question now and then with people not using OkapiLib's IMU.

I'll leave it to you to decide what to do with this issue. Feel free to close it.

WillXuCodes, 2022-08-04:
We went with adding imu_reset_blocking... thanks for the issue!