purduesigbots/pros #416: Abstract base class for motors and motor groups

Richard-Stump, 2022-09-21:
**Background**
The idea for making an abstract class for motors and motor groups to inherit ideas came up. The premise is that they have a nearly identical API and function so similarly that you should be able to treat one as the other when passing them to libraries and functions in your code.  Users could also change their motors to groups and vice versa without changing all their function declarations and implementations.

**Arguments for this feature:**
- Would allow code to use motors or motor groups interchangeably without needing to change a bunch of definitions
- Libraries can take the base class as input for motors/groups for things like chassis, odometry, etc.
- The motor groups have an API that is almost identical to the motors, so making them compatible would complete this
- We are making device configuration object oriented, so this will make 
- Would decouple libraries who need motor groups from depending on libraries like Okapi

**Arguments against this feature:**
- Libraries such as Okapi already fill this requirement
- If you change from multiple motors to a single motor, you probably have to reprogram your bot's logic/model anyway, so this feature might not add much
- The kernel should provide base functionality, and then other libraries can abstract these functionalities as they see fit
- Potentially lots of questions from users trying to instantiate the base class, or trying to do invalid conversions
- vtables might impose a small performance hit

We are looking for some more feedback on what the community thinks about this idea and any more pros/cons of this feature. If you have anything to add, please comment on this issue.

djava, 2022-09-21:
This seems like analogous to always passing motor groups because if there's only one motor in it, then there's only one motor and it's treated the same way as a singular motor anyway, so this feature isn't actually necessary for that use-case.  I see this as codifying that technique into the API in a more intuitive way though, so I'm supportive of it.

djava, 2022-09-21:
I like the idea of this feature, as it's something that is oft-implemented by PROS library authors and it's useful and not game-changing for anyone that doesn't want to use it.  However, I see the points of people who want to avoid the added tech-support burden and think the kernel should be as bare-bones as is reasonable.

I have a few options that I think are viable for PROS (most preferable first, decreasing order):
1. Create a first-party default template for `pros::AbstractMotor` (or whatever the naming scheme ends up being).  That way, people that don't want it have the option of excluding it from their project.  This wouldn't be an "abstract base class" so to speak, since `pros::Motor` and `pros::Motor_Group` can't inherit from a modularized template that isn't guaranteed to be present. However, almost all relevant functionality can be recreated using user-defined conversion functions from `pros::Motor` and `pros::Motor_Group` to implicitly convert between them, letting the end-user basically ignore that the AbstractMotor class exists, which should be the goal.

2. Something akin to
```cpp
namespace pros {
  namespace __internal {
    class AbstractMotor;
  }
  #ifdef PROS_EXPOSE_ABSTRACTMOTOR
  using __internal::AbstractMotor;
  #endif

  class pros::Motor : public __internal::AbstractMotor {
    // Motor stuff
  };
}
```
This has the benefit of being a real inherited base class, which does remove some amount of the overhead and indirection of option 1 (however trivial that would actually be in the real world).  It also gives users (and more importantly, library authors) the choice of even seeing that AbstractMotor show up in their autocomplete list, which should cut down on the tech support concerns that some have raised.  Library authors can define `PROS_EXPOSE_ABSTRACTMOTOR` if they want to use it, and it would carry all of the same benefits otherwise presented.

3. Just add it. Honestly, I don't think the questions about trying to instantiate `pros::AbstractMotor` would be that bad.  [This](https://godbolt.org/z/bTKWMTMob) is what the error looks like when you try to instantiate an abstract class.

4. Don't add it. The functionality is essentially identical to only using `pros::Motor_Group` and if you only want one motor, then only put one motor in it.  This whole thing is basically sugar and prettiness around that already natively achievable functionality.

I personally prefer option 1 because it seems to have the least drawbacks and it plays into my modular vision for pros that the dev team seems to be working towards as well (liblvgl, branchline, etc).

closed 2022-11-01 by WillXuCodes