purduesigbots/pros #423: Initializing a Rotation with any value in reverse_flag causes brain abort exception

hazardProgrammer, 2022-10-01:
Code used is the following:

```
#include main.h
pros::Rotation foo(11, true);
```

where there is a rotation sensor in port 11 (not sure what happens if there isn't). If `false` is used instead, the same thing happens.
To actually reverse a rotation sensor, I currently have to initialize it as `pros::Rotation foo(11)` and then call `foo.reverse()` in `initialize()`.

WillXuCodes, 2022-10-09:
can you send your full code? just want to see what's going on. What Kernel version are you on? I vaguely recall that we might've fixed this.

WillXuCodes, 2023-01-12:
Ok the current issue with this stands as the following: 

- We put a bandaid solution by removing a wait in the constructor. 
- However, this popped up another issue where now the reverse flag is not set automatically until a few milliseconds after initialization, so the first check of the reverse flag will always fail.

Closing this issue for now as the original issue was solved but we may need to re-visit this sometime down the line and fix it similar to the imu flag #398, by removing any scheduler calls but somehow allowing CPU 0 to breathe.

closed 2023-01-12 by WillXuCodes