purduesigbots/pros #426: PROS 4: Virtualize device classes

Richard-Stump, 2022-10-04:


Richard-Stump, 2022-11-01:
After talking with Will, we should rename the class `SmartDevice` to just `Device` since the class is already in the v5 namespace

dcieslak19973, 2022-12-13:
The `device` class defined in the VexCode API provides a `timestamp()` function which, I believe, provides the timestamp from the device's internal clock, which may drift slightly from the Brain's clock.  For precise calculations, such as those for flywheels, getting both the sensor value and the time at which that value was taken would be useful.

djava, 2022-12-13:
> * get_type(): which returns an enum identifying which type of smart device is on this port.

Do you get this from VexOS or would it be like a
```cpp
DeviceType Motor::get_type() override {
  return DeviceType::Motor;
}
```
type thing?

Richard-Stump, 2022-12-13:
> The `device` class defined in the VexCode API provides a `timestamp()` function which, I believe, provides the timestamp from the device's internal clock, which may drift slightly from the Brain's [...]

We'll look at this idea as maybe a 4.1 feature. You're right that the sensor times and the PROS time would drift over time, but it might not be too bad?

We could split this into two functions potentially: timestamp() which returns a timestamp from the PROS clock and maybe rawTimestamp() which gets the timestamp from the devices internal clock. 

The main issue I see here is the mentioned drift, and if users start to mix the two together. Is there a a specific use case you see for needing the timestamp from the device itself vs. calling `pros::millis()`?

Richard-Stump, 2022-12-13:
> 

> > * get_type(): which returns an enum identifying which type of smart device is on this port.
> 
> Do you get this from VexOS or would it be like a
> 
> ```c++
> DeviceType Motor::get_type() override {
>   return DeviceType::Motor;
> }
> ```
> 
> type thing?
It's very much like the code block you included.

djava, 2022-12-13:
What's that for then? Are we expecting it to be useful for end users? Is it useful for something internally?

dcieslak19973, 2022-12-13:
> The main issue I see here is the mentioned drift, and if users start to mix the two together. Is there a a specific use case you see for needing the timestamp from the device itself vs. calling `pros::millis()`?

I do not have direct experience with it, but based on what I've read from @SylvieMayer, there are issues with the vexos-reported velocity values coming out of motors. See: https://sylvie.fyi/sylib/docs/db/d8e/md_module_writeups__velocity__estimation.html

As well as https://github.com/SylvieMayer/sylib/blob/cb2154e7340ec6629fcd6b600d04eb46ad4322ec/src/sylib/motor.cpp#L218 which is used to compensate.

@SylvieMayer may weigh in directly with more expert knowledge/experience.

Richard-Stump, 2022-12-14:
> What's that for then? Are we expecting it to be useful for end users? Is it useful for something internally?

Yeah, it probably won't be that useful unless users want to allow any device to be passed to a function. We just thought it would be good to have it here just in case.

Richard-Stump, 2022-12-14:
> 

Alright, thanks for the insight! It would probably be good to have this then. I'll ask them for some more insight.

nickmertin, 2022-12-14:
> The main issue I see here is the mentioned drift, and if users start to mix the two together. Is there a a specific use case you see for needing the timestamp from the device itself vs. calling `pros::millis()`?

In my opinion I don't think clock drift is the particular issue this will be useful for. I think the issue is not knowing when in the last x milliseconds the measurement was taken. Particularly when manual discrete differentiation is needed/desired, having a decently accurate clock for the sensor readings - even if it isn't synchronized with the main system clock - would be very useful in improving the accuracy of the estimate.

closed 2023-02-09 by Richard-Stump