purduesigbots/pros #456: PROS 4: Change any "reset" devices to "tare" (RFC)

WillXuCodes, 2022-10-31:
#### Expected Behavior:
In some cases, our API has "reset_position" or a "reset" to indicate that we are setting the current position as 0. This however is inconsistent with other parts of the API which have a "reset" to reset the device and put it into a calibration mode.

At the moment, I see that we have 2 options to go about fixing this:
1. Change the functions that zero the devices as "tare" and keep the calibration mode functions as tare.
2. Change the functions that reset into calibration mode to ".calibrate" and change the tare functions to "reset".

Thoughts?

Richard-Stump, 2022-10-31:
I think that `reset()` is a poor name choice for the calibration mode functions, and `calibrate()` would be better since it is more explicit. As for the `reset_*()` functions, we could probably leave them as `reset_*()` if we change the calibrate functions to have a better name, but tare would probably be more accurate to what those functions do (and it is the more scientific term).

WillXuCodes, 2022-10-31:
I'm not sure if I agree with changing reset to calibrate to be honest, since it's already convention to use "reset" at the beginning to calibrate devices. However, we could also just add tare_position() as an alias to reset_position()... it's only an issue in the rotation sensor if I'm going to be honest and we could just fix it there and asign that task to one of the newer members.

closed 2023-03-08 by WillXuCodes