purduesigbots/pros #471: ✨Method to determine Field Control from Competition Switch

ifndef-define, 2022-11-26:


Jerrylum, 2022-12-22:
Not sure why no one answered you, you can implement this by using `((pros::c::competition_get_status() & (1<<3)) != 0)`.

djava, 2022-12-22:
That may be worth adding in for PROS 4, it seems useful and is not an intuitive implementation for users (actually looks like that's entirely undocumented, so it should at the very least get docs).

ifndef-define, 2023-01-04:
Docs would probably be enough since it isn't widely used. I was searching through the website and couldn't find anything about this conditional check. Thanks for letting me know!

WillXuCodes, 2023-01-12:
Please refer to this issue for the implementation of whatever we choose to do for our API in PROS 4. 

https://github.com/purduesigbots/pros/issues/102

PR #608 (2023-10-27): ✨Field Control State Getter
#### Summary:

Added two functions to check if the brain is connected to field control or competition switch.

#### Motivation:

##### References (optional):

closes #471 

#### Test Plan:

connect the brain to competition switch/field control and call the two functions to see if they return the correct value
closed 2024-02-09 by CChheerryyll