purduesigbots/pros #497: Add a "is_installed()" method for devices

ifndef-define, 2023-01-25:
While reading the documentation, I found no dedicated method to check whether a V5 device (motor, distance, radio, etc.) is installed. While this isn't something major since you could just call any method and check for the return of an `errno` of `ENODEV`, I thought it might be worth adding a simple check method like `MOTOR.is_connected()` or `MOTOR.is_installed()`.

Richard-Stump, 2023-01-25:
I think there's been little reason for such a method since most of the time, teams plug their devices in and the code works because they are there. There is little in the way of connecting/disconnecting devices except for building the robot. However, this would be useful for sanity checks and would be clearer than checking errno.

Additionally, would it be useful to introduce a standalone function that can check if there is a device on a port, and if so return the type of device connected?

I'm marking this as RFC (Request For Comment) for the time being to get some more feedback on this idea.

ifndef-define, 2023-01-25:
Yea like I said, such a method is there for the sake of it being present. Like a program that warns the driver in case a vital motor disconnects in game or something. I was thinking of having a way to manipulate a 6-motor drive so like should a motor on the right side disconnect, the program can detect that and reduce left side power to maintain a straight tracking. 

> Additionally, would it be useful to introduce a standalone function that can check if there is a device on a port, and if so return the type of device connected?

I don't think this kind of feature is needed. At least I can't think of a way this could be utilized. This would probably be useful if the robots could change device ports automatically.

closed 2023-03-19 by WillXuCodes