purduesigbots/pros #508: Color Enum does not work properly with motor gears.

noam987, 2023-02-03:
The pros::Color enum is currently being static casted to motor_gear enum class, but the values don't match up leading to bugs.

noam987, 2023-02-03:
Example: 
pros::Motor_Group left({1, 2});

left.set_gearing(pros::v5::Motor_Gear::green);
left.get_gearing(); // returns 1 = green

left.set_gearing(pros::Color::green);
left.get_gearing(); // returns 0x00008000

ryan-doan, 2023-02-04:
I'm thinking we can have an if statement in motor_set_gearing to check if we're using pros::Color, and then a switch statement to convert the enum to pros::v5::Motor_Gear. Would that suffice?

terrapinrobotics, 2023-02-07:
Looks like this functionality is going to be removed in PROS 4, so these changes may not be necessary.