purduesigbots/pros #525: RFC: PROS 4.0.1 Motors/Motor Group Design

Richard-Stump, 2023-02-10:
### Background
The current design for Motors, Motor Groups, and their relation is not well liked and goes against C++ norms. We are reverting back to the original idea since the better maintainability does not out weigh the priority of a good API. 

### Proposal:
Motors and motor groups inherit from a class named `AbstractMotor` like originally planned. This class is not constructable (as it is abstract) and contains virtual methods for functions from motors and motor groups
- The motor's move functions. For motor groups, they tell all the motors in the group to move uniformly
- The motor's set functions. For groups, they set the parameter for each motor uniformly
- The motor's get functions: For groups, they get the value from the first motor
- The motor group's get_all_* functions. For motors, they return a vector of 1.

We are also removing the OOP features from the motors/groups since they caused too many issues. The constant calls to send the parameters caused issues. We're moving the OOP reversed flag to the C api level. 

Looking for any more thoughts and considerations on this.

djava, 2023-02-10:
This sounds much nicer, I appreciate this decision 👍👍

Richard-Stump, 2023-02-23:
Closed since #532 implements this.

PR #532 (2023-02-23): ✨PROS4: Base Class for Motors
#### Summary:
Reworks the inheritance between motors and motor groups by adding a new BaseMotor virtual class. 

#### Motivation:
#525 

##### References (optional):
Closes #525 
Closes #507 by deciding that this should NOT be intended behavior
Closes #490 as it doesn't make sense in the current architecture. Motor groups are vectors of ports, not of motor objects. We have added indexable methods for all non-movement methods to target any one specific motor in a motor group.

#### Test Plan:
- [ ] It compiles
- [ ] Make a motor and motor group and test out some me [... cut]
closed 2023-02-23 by Richard-Stump