purduesigbots/pros #526: ADIEncoder construction with 3-wire expander port fails without specifying reversal

SizzinSeal, 2023-02-11:
#### Expected Behavior:
An ADIEncoder should be able to be constructed without giving a value for whether it should be reversed or not.

#### Actual Behavior:
When passing a tuple for the port (when the user uses a 3-wire expander), the user has to pass a boolean as well for specifying whether the encoder should be reversed or not. Otherwise it does not build.

#### Steps to reproduce:
Below is some example code
```c++
pros::ADIEncoder enc('A', 'B');  // builds
pros::ADIEncoder enc('A', 'B', false); // builds
pros::ADIEncoder enc({1, 'A', 'B'}); // does not build
pros::ADIEncoder enc({1, 'A', 'B'}, false); // builds
```

#### System information:
Platform: V5
PROS Kernel Version: 3.7.3

#### Screenshots/Output Dumps/Stack Traces
```
src/main.cpp:16:43: error: call of overloaded 'ADIEncoder(<brace-enclosed initializer list>)' is ambiguous
   16 | pros::ADIEncoder verticalEnc({7, 'A', 'B'}make: *** [common.mk:270: bin/main.cpp.o] Error 1
);
      |                                           ^
In file included from ./include/api.h:65,
                 from ERROR - pros.cli.build:make - Failed to make project: Exit Code 2 - pros-cli version:3.4.1
PROS-CLI Version:  3.4.1
./include/main.h:37,
                 from src/main.cpp:1:
./include/pros/adi.hpp:499:2: note: candidate: 'pros::ADIEncoder::ADIEncoder(pros::ext_adi_port_tuple_t, bool)PROS-Kernel Version: 3.7.3
'
  499 |  Error: Failed to build
ADIEncoder(ext_adi_port_tuple_t port_tuple, bool reversed = false);
      |  ^~~~~~~~~~
./include/pros/adi.hpp:465:7: note: candidate: 'constexpr pros::ADIEncoder::ADIEncoder(const pros::ADIEncoder&)'
  465 | class ADIEncoder : private ADIPort {
      |       ^~~~~~~~~~
./include/pros/adi.hpp:465:7: note: candidate: 'constexpr pros::ADIEncoder::ADIEncoder(pros::ADIEncoder&&)'
```

djava, 2023-02-11:
Somewhat poorly worded issue, he just wants the ADIEncoder ctor to have a default value for the reversed parameter.

Richard-Stump, 2023-02-11:
> Somewhat poorly worded issue, looks like he just wants the ADIEncoder ctor to have a default value for the reversed parameter.

What they're after is perfectly clear from their description.

SizzinSeal, 2023-02-11:
> Somewhat poorly worded issue, looks like he just wants the ADIEncoder ctor to have a default value for the reversed parameter.

It does have a default parameter and yet it does not build
ADI.hpp, line 499:
`ADIEncoder(ext_adi_port_tuple_t port_tuple, bool reversed = false);`

Richard-Stump, 2023-02-11:
Oh, I see. We have multiple ctors and C++ can't decide which one is right for your case. I'm guessing that it can't be fixed in PROS 3, but we'll take a look.

WillXuCodes, 2023-02-22:
Fixed! Thank you for the issue.

PR #533 (2023-02-22): :bug: PROS 3.8: ADIEncoder Ambiguous Default Constructor
#### Summary:
Closes:
#526 

Makes it unambigious by deleting the copy constructor for ADIEncoder. While this is a breaking change this should not really effect many codebases that attempt to do this.  

#### Test Plan:

[image]

Does not break ABI for any of these major templates.
closed 2023-02-22 by WillXuCodes