purduesigbots/pros #556: PROS 4: Update structure/class/enum case sytles

Richard-Stump, 2023-03-13:
### Description:
Currently, there's some structures/classes/enums that use underscores in their naming convention, such as `Motor_Group` and `Motor_Gearset` to name a couple. 

It would be nice if we could get these updated to be in pascal case like this: `MotorGroup`, `MotorGearset`, etc. since we have the ability to make the breaking changes. This has been requested by a few users in the past, and I think it looks slightly cleaner personally.

### RFC Items:
* Should we keep the old names as aliases to avoid breaking so much code?
* Should we not so that users don't use the old names and we don't have to maintain the aliases? (It's not a huge deal to maintain them IMO)

Richard-Stump, 2023-03-13:
Another note: If we do this, we would want to go through and really make sure that the styling on the names is consistent since this has been an issue in the past IIRC.

djava, 2023-03-13:
> Should we keep the old names as aliases to avoid breaking so much code?
fwiw pros 2->3 changed the casing style of every function name from camelCase to snake_case

djava, 2023-03-13:
Also yeah I'd definitely like to see PascalCase for PROS 4. I'm split on keeping aliases.

What if we ship PROS 4 with a python migration script to auto-convert all the Upper_Snake_Case to PascalCase (half joke).

WillXuCodes, 2023-03-19:
Honestly I'd rather have PascalCase, just looks cleaner.

closed 2023-03-22 by WillXuCodes