purduesigbots/pros #592: Copy operator for motor_group implicitly deleted

AvereyTheTerrible, 2023-08-26:
I get this error when I try to assign a motor_group to another motor_group. It seems improbable that this is an intended behavior of motor groups
#### System information:
Platform: 
PROS Kernel Version: 3.8.0

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
[image]

AvereyTheTerrible, 2023-08-26:
On further inspection this seems to be because mutexes have explicitly deleted copy operators. If this is intentionally maybe explicitly delete the copy operators for motor_groups?

djava, 2023-08-26:
yeah this seems surely unintentional. Easy fix to define a custom copy ctor (and move) ctor.

AvereyTheTerrible, 2023-08-26:
I don't think that will work since it would still try to copy mutexes which have explicitly deleted copy operators. I'm no expert though.

djava, 2023-08-26:
It wouldn't unless you told it too. You'd only copy the ports vector in that copy ctor.

AvereyTheTerrible, 2023-08-27:
Oh yea I guess that works. The copied motor_group would kind of become a reference then.

closed 2023-12-13 by WillXuCodes