purduesigbots/pros #593: ✨ Remove Motors from Motor Groups by Port

SizzinSeal, 2023-08-26:
#### Requested Feature:
Users should be able to remove motors from motor groups by specifying the port of the motor to be removed.

#### Motivation:
Power sharing mechanisms such as ratchets and PTOs can be complicated to program, as multiple subsystems could use said motor but only one can control it at a time. Having the ability to remove a motor from the motor group by specifying its port can make this significantly easier, especially if the user is also using a template which controls the motor too.

#### Example:
A robot has 4 motors on its drivetrain. 2 of these motors are connected to a PTO. The user uses a library to control the drivetrain, but needs to have control over the 2 motors when the PTO is engaged. In order to do this, the user would remove the motors from their motor group:

```cpp
// remove motor on port 1 from motor_group
motor_group.remove(1);
```

#### Additional Information:
* Users should be able to remove motors by specifying their port or index
* If the user commands a motor not part of the motor group to be removed a runtime error should be thrown

djava, 2023-08-26:
I'd honestly probably just recommend having multiple motor groups to represent the pto on/off.

I can't think of much reason you'd need to dynamically change the motors in a group, you should probably always know the various states statically.

SizzinSeal, 2023-08-27:
> I can't think of much reason you'd need to dynamically change the motors in a group, you should probably always know the various states statically. 

I disagree. By moving the motor from one motor group to another makes controlling what has access to the motor easy without the need for race conditions etc.

noam987, 2023-08-30:
This behavior exists for PROS 4 using Motor_Group::erase_port(int). We currently don't have plans to backmerge this into PROS 3, due to the imminent release of PROS 4. However we should look into renaming this method as I agree the naming is unclear

closed 2023-08-30 by noam987