purduesigbots/pros #598: ✨[FEATURE REQUEST] Update LVGL

kisvegabor, 2023-09-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Hey,

It's Gabor from LVGL. Recently I see more and more LVGL users from PROS, and I noticed that you are still using a [very old](https://github.com/purduesigbots/pros/blob/develop/include/display/lv_version.h#L17) version of LVGL. I wonder if you are interested in updating to v8. If so, I would be happy to help with any LVGL related questions. 

**What is the behavior that isn't addressed by the existing API?**
N/A

**OPTIONAL Describe any solutions you've considered**
Update to LVGL v8 :slightly_smiling_face: 

**Additional context**
We will release LVGL v9 in by the end of this year, but updating to v8 would be already huge step.

BennyBot, 2023-09-24:
In our new major version, PROS Kernel 4, we have updated to LVGL version 8.3. Additionally in this version, we have decoupled lvgl from our kernel into a separate template, liblvgl (https://github.com/purduesigbots/liblvgl), to allow us to support multiple versions and upgrade versions easier. We are currently supporting lvgl 5.3 for a future kernel for those who do not want to have to rewrite their existing lvgl code when they upgrade, while also supporting lvgl 8.3 for those who wish to use new features, emulators, etc. We aim to have this version out as a public open beta by the end of October, and have it be the new default version at the end of this years VEX Competition Season (end of April 2024). While we would like to make this the default version ASAP, there are many breaking changes between the current kernel version and PROS Kernel 4 that would not be good to introduce in the middle of a competition season.

We will look at supporting LVGL v9 as well, thank you for this information.

kisvegabor, 2023-09-25:
Thank you for the summary, it sounds great! :slightly_smiling_face: 

Let me know you need any help from my side. If you wish we can organize a meeting as well, just to get to know each other and explore some cooperation opportunities.

closed 2023-09-27 by BennyBot