purduesigbots/pros #673: 🐛[BUG] - Inconsistencies in some llemu functions requiring liblvgl while others don't

Unionjackjz1, 2024-06-03:
### Describe the bug
Some (seemingly) missing llemu functions throw errors without liblvgl 8.3.8, while other functions don't seem to require it.  

### To Reproduce
Steps to reproduce the behavior on a fresh project:
1. Create a new project
2. In `initialize()`, replace `pros::lcd::register_btn1_cb(on_center_button);` with `btn0` or `btn2`.  
3. Build the project.  You will get `error: 'register_btn2_cb' is not a member of 'pros::lcd'`
4. Apply `liblvgl@8.3.8`
5. Build the project and everything works as normal
6. Make a template that uses `btn0` or `btn2`
7. Applying this to a project without liblvgl will throw errors.  Applying this to a project with liblvgl will work.  

This can be recreated in [release/3.1.0 of EZ-Template](https://github.com/EZ-Robotics/EZ-Template/tree/release/3.1.0).  The base project has liblvgl installed and EZ-Template-Example-Project does not.  Removing liblvgl from the main project throws errors, adding liblvgl to the example project fixes the errors. 

### Versions of stuff
 - Windows
 - okapilib 5.0.0
 - liblvgl 8.3.8
 - EZ-Template 3.1.0
 - kernel 4.1.0

### Additional Information
These are the functions I see this is happening with, and all aren't in `llemu.hpp`.  There could be more, and this could be related to #672? 
- clear()
- shutdown()
- register_btn0_cb()
- register_btn2_cb()

WillXuCodes, 2024-06-04:
So this issue is multifaceted:

On the kernel side of this that could be an enhancement:
- This is a side effect of us declaring main.cpp LLEMU calls as weak symbols so it compiles without liblvgl installed.
- Maybe we should just declare all LLEMU calls as weak symbols so user code can be agnostic to if LVGL is installed or not in reference to LLEMU? 

The actual bug part of this: 
- `liblvgl` is not being installed by default by the CLI. This is a major issue as lvgl is an integral part of using the screen with PROS.

hpyhacking, 2024-06-04:
thanks for this thread I just figure out why the screen black and lcd initial failed.

Just `pros c apply liblvgl `