purduesigbots/pros #676: ✨[FEATURE REQUEST] Async Compiling

SizzinSeal, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
Building large projects from a clean build can take a long time. This mostly affects automatics like a GH workflow that compiles the project, or an azure pipeline that builds the project. While not exactly an issue, it would be great for QoL.

**Additional context**
running `make quick -j` does compile the project asynchrononously but is unreliable

djava, 2024-06-05:
Uh, what exactly do you mean by this? You can't compile asynchronously, it has to be done before the pipeline is done, obviously...

Do you mean in parallel? `make -j$(nproc)` is the typical way to do this, what do you find unreliable about it? This is almost certainly not a pros issue.

nickmertin, 2024-06-05:
I second that "unreliable" is not exactly actionable information. Is it sometimes failing to compile? Giving incorrect output? Sometimes not actually executing in parallel?

SizzinSeal, 2024-06-05:
ok i can't seem to recreate it now? anyway yes sorry i meant compile in parallel, not asynchronously

closed 2024-06-05 by SizzinSeal