purduesigbots/pros #681: Motor Configurations use negative ports for reversed motors

D-Colak-PC, 2024-06-23:
**Is your feature request related to a problem? Please describe.**
When creating motor objects, it is possible to indicate a reversed motor by writing a negative port number. This syntax doesn't make sense since there are no negative ports on the VEX Brain.

**Possible Solutions**
Force the user to use the `reverse` boolean when defining motors and remove negative ports.

**Current Syntax**
user can chose between
`pros::Motor motor(-PORT) // reversed port number`, and
`pros::Motor motor(PORT, true) // reversed bool = true`

**Solution**
Force the user to use
`pros::Motor motor(PORT, true) // reversed bool = true`

D-Colak-PC, 2024-06-23:
meant to label this as bug, sorry

djava, 2024-06-23:
Can we remove the support for the IMU while we're at it? I don't really like the IMU, so I don't want to use it.

djava, 2024-06-23:
Anyway, no. Don't use the feature if you don't want to, it's very much not required.

D-Colak-PC, 2024-06-23:
Thanks for the feedback...
I don't know if the sentiment is shared by others using this tool, but using the sign of a port number to indicate the direction of the motor can be unintuitive, especially for new users. Using negative ports coupled with the reverse flag can cause some confusion as well. It's not entirely clear what this motor's direction is: `pros::Motor motor(-PORT, true)` without looking into the PROS source code.
Right now there are two ways to achieve the same trivial task, with one objectively more confusing and unintuitive than the other. I believe any step we can do to streamline the process to make it as easy as possible to work with this tool should be taken, and removing unintuitive configuration options can help with that.

aberiggs, 2024-06-23:
Just want to point out that the removal of any function/feature will result in the failure to compile any pre-existing projects that teams have built. IMHO I think it only makes sense to leave both methods in and let users choose what they'd like to use. As said above, nobody has to specify negative port numbers to reverse motors if they don't want to, but for those who prefer that syntax, forcing them to move to what to you feels "more clear" is problematic. It being "objectively confusing" doesn't seem to me a strong enough of a reason to make a breaking change.

noam987, 2024-06-23:
>It's not entirely clear what this motor's direction is: `pros::Motor motor(-PORT, true)` 

I would like to point out that this is no longer valid syntax for pros 4. there is no boolean for setting reversed in the constructor, mainly due to it being to confusing in exactly this case. You still have the `motor.set_reversed(bool)` function if you would like to set reversal of a motor without using negative ports. 

This was an intentional change in pros 4.1 in part to reduce the number of arguments in motor constructors and to be more resilient to motors getting unplugged in the middle of programs.

We do really appreciate the user feedback, and continue to welcome any more comments you have when using pros 4.1.

D-Colak-PC, 2024-06-23:
Sorry if I came across negatively...

> Just want to point out that the removal of any function/feature will result in the failure to compile any pre-existing projects that teams have built.

I hadn't thought of this, and it makes much more sense to me now. It's definitely not worth forcing hundreds of projects to change because of a minor syntax change would not be a good idea.
> I would like to point out that this is no longer valid syntax for pros 4.

I must have been looking at the wrong version... silly me 😅. I'm still learning the ropes of github and contributing to open source, so I apologize for any inconvenience or time-wasting I may have caused. I appreciate the feedback though, and I'll be more careful before opening issues or PRs in the future.

djava, 2024-06-24:
Generally, removing something from an API requires more of a "this is wrong if you use it" justification. Anyway, this is a widely used and appreciated feature, and is quite a bit clearer than just having an unlabeled `true` after the number in the definition IMO.

closed 2024-06-24 by D-Colak-PC