purduesigbots/pros #705: ✨[FEATURE REQUEST] C++ Module Support

RandomPythonProgrammer, 2024-10-12:
I would like to use the Modules feature from C++20 in my project [cppreference](https://en.cppreference.com/w/cpp/language/modules).

When I currently compile I get this error:
`warning: keyword 'export' not implemented, and will be ignored`
I have tried adding this compiler flag:
`-fmodules-ts`
But the flag is not recognized.

I am not sure if this is a bug or an unimplemented feature because I know support for modules took a while for compiles to add. Thank you for your hard work in maintaining pros and keeping it free and open source.

AndrewLuGit, 2024-10-12:
The current PROS toolchain uses GCC version 10. Support for modules was not added until GCC 11. We are planning to update to GCC 13 in the next update of PROS.

djava, 2024-10-12:
It's unlikely that you'd have a good time trying to use modules using PROS's build system (Make). Modules require a pretty fancy build system, such as CMake, and Make is not really capable of doing all the work necessary for modules to compile nicely.

thiccaxe, 2024-10-12:
Makefiles are generally custom-made for a project, whereas CMake generates Makefiles automatically using a configuration language.

djava, 2024-10-12:
The PROS Makefile was written by the PROS team, and is used by all PROS projects

RandomPythonProgrammer, 2024-10-14:
Okay thank you so much for the insights!

closed 2024-10-14 by RandomPythonProgrammer