purduesigbots/pros #733: :sparkles: [FEATURE REQUEST] Hardware Accelerated Square Root

SizzinSeal, 2024-12-24:
#### Requested Feature
Hardware accelerated square root. The Cortex A9 on the V5 Brain has instructions to compute the square root of floating-point numbers through the [VSQRT](https://developer.arm.com/documentation/100076/0200/a32-t32-instruction-set-reference/floating-point-instructions--32-bit-/vsqrt) instruction.

#### Current Implementation
Currently, GCC uses the sqrt implementation of libm which does not make use of this instruction.

#### Motivation
Computing the square root of a floating-point number is an expensive operation. With the use of more expensive algorithms like live-regenerating motion profiling and Monte Carlo Localization on the rise in VRC, it makes sense to get the V5 to perform as well as it possibly can. The better it performs, the less PROS users have to worry about performance limitations.

#### Features of `VSQRT`
- Supports single-precision and double-precision floating-point numbers
- Can produce "Invalid Operation" or "Inexact" exceptions (e.g, square root of -1, square root of 1 Googol)

#### Why doesn't GCC use this instruction automatically?
libm uses errno. Yup, that simple. GCC can't inline VSQRT because it does not set errno. The solution is to use `-fno-math-errno`, but this is unfortunately a breaking change. However, the chances of someone checking errno after running libm functions are very, very slim.

djava, 2024-12-24:
You're asking to set `-fno-math-errno`? No.

Exortions, 2024-12-24:
> You're asking to set `-fno-math-errno` in PROS? No.

Why not? Does pros use errno in a lot of places?

SizzinSeal, 2024-12-24:
PROS doesn't but something like Eigen might. I've discussed this with @djava and he's convinced me this is not the route to take

Exortions, 2024-12-26:
👍

SizzinSeal, 2024-12-26:
@Rocky14683 has this been completed? I didn't see a commit to PROS which implements this

Rocky14683, 2024-12-26:
i thought u said its not the route to take

SizzinSeal, 2024-12-26:
Sorry, I should have been more specific. While we can't use `-fno-math-errno`, we can implement our own `sqrt` operation which uses the `VSQRT` instruction, but also sets errno

ion098, 2024-12-26:
Re @SizzinSeal 

> While we can't use `-fno-math-errno`, we can implement our own `sqrt` operation which uses the `VSQRT` instruction, but also sets errno, so behavior is still the same as previously, just much faster

Doesn't newlib already do this? https://github.com/bminor/newlib/blob/363357c023ce01e936bdaedf0f479292a8fa4e0f/newlib/libm/machine/arm/e_sqrt.c#L30-L41

SizzinSeal, 2024-12-31:
Perhaps it does after the libm update PR. I've only analyzed binaries created using PROS 4.1.1, I'll find out