purduesigbots/pros #769: libgcc.a uses VFP register arguments, bin/cold.package.elf does not (hard-float upgrade fails)

jpalardy62, 2025-04-15:
Testing with kernel 4.2.0, vscode addon version 0.8.3 (released yesterday), GCC `arm-none-eabi-gcc (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614` on macOS (M1 / Apple Silicon)

All of these versions * should * be correct, per #703 , but it's still failing to compile. Downgrading to kernel 4.1.2 (`pros conduct upgrade kernel@4.1.2`) compiles correctly, upgrading back to 4.2.0 fails again. I've also uninstalled/reinstalled the toolchain.

AndrewLuGit, 2025-04-15:
Try running `pros make clean` to delete the cold package, then see if it build properly

apalrd, 2025-04-16:
Already did that as well.

djava, 2025-07-08:
It would be pretty shocking if a python version change breaks `pros make` a little but not entirely... `pros make` (and `pros make clean`) just directly call `make` and, afaik, do nothing else